muxall/RPTCQ-LCD #1: Max time on profile 995 disappears.

muxall, 2020-02-02:
Carter: found one more repeatable bug.. if you set the time to max on a profile at 995 hit save, then leave that profile and come back the timer disappears, restarting it was the cure as no cursor or numbers appeared...i repeated this on profile 3 as well with the same result

kennethLeeHarris, 2020-06-18:
I tested it out came across the same problem. When I put 995 hit save it restarts to the default settings. I try out profile 1 and 4.

muxall, 2020-06-19:
Fixed in 0.450.
Fix was applied to profiles 1-4.  When testing, verify all 4 profiles max out at 995.
Also, verify the web app will not allow setting the time above 995 for all four profiles.

kennethLeeHarris, 2020-06-22:
I tested it out on version 0.450s. All four of the profiles worked on 995. When you try to go above 995 after you click Enter/Save it defaults to 995.

closed 2020-06-23 by muxall