muxall/RPTCQ-LCD #14: Feature: Add an alert to web app if grill is alarming from flame-out or some other cause.

muxall, 2020-03-27:
Feature: Add an alert to web app if grill is alarming from flame-out or some other cause.

muxall, 2020-06-17:
Here is a javascript solution.  I'm not sure we have enough memory for it.
var snd = new Audio("beep.wav"); // buffers automatically when created
snd.play();

muxall, 2020-06-19:
The web app should beep when:
1. 4x The grill changes state to cool down.
2. 4x The grill changes state to off.
3. 1x The ADC is reset.
4. 4x The grill flame-out detection changes to Pause state.

muxall, 2020-06-19:
Flame-out detect:
1. Set Flame-out detect time to 5 minutes; click save.
2. Set Profile 1 to 225. Un-check Hold.
3. Set Profile 2 to 180. Check Hold.
4. Click Enter/Save.
5. Click Start.
The flame-out detect should alarm when dropping from 225 to 180.

kennethLeeHarris, 2020-06-19:
On the flame-out detection change to pause state. It did not beep four times. It beeped every other second to I hit the start button. For cool down it did beep four times before starting.

closed 2020-06-29 by muxall