muxall/RPTCQ-LCD #18: Issue: Add a check to see if a BBQ cook is running before proceeding  with update.

muxall, 2020-04-18:
Add a check to see if a BBQ cook is running before proceeding  with update.

muxall, 2020-07-29:
Fixed in 0.505

kennethLeeHarris, 2020-07-31:
I wasn't sure what add a check to see if barbecue cook is still running before processing with updated. So what I did was made a cook then I updated the controller to see if anything would come up. All that came up was the warning that I would lose my recipes. Then I hit okay it updated the controller and stop my cook. I need to talk to you before changing the milestone.

muxall, 2020-08-01:
You were the one that found this bug.  Replace the simulator load with the real 0.506 load and then hit start.  Since the simulator won't be running, it will never get above 80F and will fail Flame-On detection.

muxall, 2020-08-01:
Fixed, again, in 0.507.

kennethLeeHarris, 2020-08-03:
Tested it on 0.507s work great. When I try to update it the Web App said updates are only allowed when barbecue state is off.

closed 2021-02-10 by muxall