muxall/RPTCQ-LCD #40: Issue: Probe #4 is allowed in  Cook-By-Probe  from the Status screen.

muxall, 2020-06-16:
From the front panel Status screen, using the cursor, select the Cook-By-Probe Probe number.  Then, using the knob, scroll up the probe numbers.  It should stop at probe #3 but it allows probe #4 to be selected.

muxall, 2020-06-18:
Fixed in 0.450

muxall, 2020-06-19:
To test:
1. on the LCD status screen change the probe number and/or the targetProbeTemp.  Verify the web app is updated with the changes.
2. On the LCD EndCtl screen, verify that changing Cook-By-Probe, Probe Number and target probe temp, and predefined target probe temps are updated in the web app.
3. On the web app, verify that changes made to Cook-By-Probe settings are updated on the LCD Status screen and EndCtl screen.

kennethLeeHarris, 2020-06-22:
1. I tested out both the Probe Number and the Probe Target Temp. If you change it on the LCD screen it will update in the web app in Cook Settings but not in Cook State.

2.?

3. I tested Cook-By-Probe settings out on the web app and the settings do updated on the LCD status screen.

4. Also I like the Max #prb reached!

closed 2020-06-24 by muxall