muxall/RPTCQ-LCD #73: Issue: Detect flame

kennethLeeHarris, 2020-07-02:
Detection Enable Flame-Out
If the target temp is same as outside temp

muxall, 2020-07-17:
Fixed in 0.504 and enhanced.
Flame-Out detection tests for three things:
1. Is Ignite Fire successful?  This is determined by a 25F rise in temp above the Outside Temperature.  Note – If your grill is in direct sunlight or has been previously used and is still hot, these may give a give a false positive to this test.  
2. Is the grill deliberately dropping temp?  Example: Going from 300F in Profile 1 to 180F in Profile 2 is a deliberate drop in temp.
3.  How long the auger has been running with the chamber temp lower than or equal to the target temp?  And, is the chamber temp climbing, dropping or  staying steady?  The Flame-Out Detect Time(m) is a timer that sets how long the grill is allowed to drop temperature (with the auger running) before it shuts the grill off.

kennethLeeHarris, 2020-07-31:
I think I might need the real grill. I look at new manual and I put profile 1 at 300 for 10 minutes and then made profile 2 180 and put it on hold but the chamber temp just caught up with it 28 minutes later. I'm going to need to talk to you to see if I did this right.

muxall, 2020-08-01:
You were the one that found this bug. Replace the simulator load with the real 0.506 load and then hit start. Since the simulator won't be running, it will never get above 80F and will fail Flame-On detection.

kennethLeeHarris, 2020-08-05:
Tested it on 0.506 work great. It pause right after Igniter Fire mode.

closed 2021-02-10 by muxall